andi34/photobooth #367: [BUG]: Can not write config/my.config.inc.php after initial install with installer script

Raukze, 2022-01-14:
### Describe the bug

After the initial install with the raspbian installer script I get `Abort. Can not write config/my.config.inc.php.` when I open the webpage. This seems to be due to the owner of the file `my.config.inc.php` being `root:root`.

### Steps to reproduce the behavior

1. Run the installer script with installing latest stable build option

### Expected behavior

Show the webpage to be able to take pictures

### Photobooth Version

latest stable from installer script

### Did it work before?

Yes

### Latest working version

-

### Additional context

`sudo chown www-data:www-data /var/www/html/config/my.config.inc.php` fixes the problem

andi34, 2022-01-14:
You've enabled Pi Camera via Installer?

andi34, 2022-01-14:
https://github.com/andi34/photobooth/commit/a2817a98e5d2cc3ed421a3eae20bfe49948fb428

If yes, this should fix the issue.

Raukze, 2022-01-15:
> You've enabled Pi Camera via Installer?

Yes I have, but as stated [here](https://forums.raspberrypi.com/viewtopic.php?t=323462) the Pi Cam does not work on Raspberry Pi OS Version Bullseye. But that's another problem...

I just tested it again with the updated script and it now sets the ownership correctly. Thanks for the quick fix!

closed 2022-01-15 by Raukze

andi34, 2022-01-15:
@Raukze you can enable the legacy camera Interface and use `raspistill` instead `libcamera-still`.

andi34, 2022-01-15:
See https://github.com/andreknieriem/photobooth/issues/296#issuecomment-1008357806