andi34/photobooth #370: [FEATURE]: virtual keyboard for enter e-mail addresses

tembelten, 2022-01-22:
### Is your feature request related to a problem?

No

### Description

With just a touchscreen it's not possible to enter an e-mail address.

### Describe the solution you'd like

it would be nice if a virtiual keyboard opens when you click on the text field and closes when pushing enter.

### Describe alternatives you've considered

-

andi34, 2022-01-22:
This is a limitation of Pi OS and not sure if it's really needed to be implemented.

https://github.com/hodgef/simple-keyboard
Might be something someone could implement, but should be optional via config. I'd be happy to give feedback on a Pull request adding this option.
Currently other stuff is on my personal To-do-List before I would look at it.

tembelten, 2022-01-28:
Hallo Andi, 
ich habe mittlerweile selbst, mit Hilfe von [https://www.youtube.com/watch?v=N3cq0BHDMOY](url), ein Virtuelles Keyboard erstellt. 

Das ganze funktioniert auch soweit wenn es über meine **index.html** aufgerufen wird.:
`<input class="mail-form-input" style="position: absolute; top: 130px; right: 30px; width: 300px;">`

Jedoch kriege ich es nicht in deiner **index.php** hin.. 
Hier lautet der Aufruf folgendermaßen:
`<input class="mail-form-input" size="35" type="email" name="sendTo">`

Die Abfrage in meinem **Keyboard.js** sieht so aus und funktioniert auch über den Aufruf über meine index.html:
`document.querySelectorAll(".mail-form-input").forEach(element => {
            element.addEventListener("focus", () => {
                this.open(element.value, currentValue => {
                    element.value = currentValue;
                });
            });
        });`

Kannst du mir weiterhelfen? Ich verstehe nicht wieso es nicht funktioniert.
Ich finde das Mailfeature wirklich spitze und fände es Schade wenn es an der Eingabe an meinem Touchdisplay scheitert..

tembelten, 2022-01-31:
Ich habe den Fehler gefunden. Das Keyboard öffnete, alledings im Hintergrund.
Musste also nur den "z-index" anpassen.
Ich lade demnächst eine genaue Anleitung hoch.

andi34, 2022-01-31:
Das wäre super, ist bestimmt für den ein oder anderen interessant. Du kannst auch einfach die geänderten bzw neuen Dateien hier anhängen, oder committen und auf deinen GitHub fork hochladen.

Hast du es auch mit dem `simple-keyboard` versucht? Das könnte man recht einfach über yarn mit installieren.
Ich bin noch nicht dazu gekommen mir das ganze anzuschauen.

mjmdurand, 2022-02-07:
hi, i made this improvement on mine and here's the steps : 
`sudo apt install onboard`
- go to onboard parameters (main menu > preferences > onboard parameters)
-  in general tab : check "auto show when editing text" and uncheck "show floating icon"
- in windows tab, check "dock to screen edge" and set it as you prefer
- in layout tab, choose "phone"
- in theme tab, cheese a theme (backboard or nighshade are good ones)
- in auto show tab, uncheck "hide when typing on a physical keyboard"

you can see all these steps here : https://www.youtube.com/watch?v=EmQR9z_krMo&t=330s

in addition to this, i advice you to set a black color wallpaper on your rpi because when keyboard appears, you can seet it behind ;D

daveschafer, 2022-02-27:
Ich habe noch eine alternative gefunden, die Chrome Extension "Virtual Keyboard". Das funktioniert fast einwandfrei, nur aus der Galerie nicht (weil die Bildvorschau das Keyboard überdeckt).

https://chrome.google.com/webstore/detail/virtual-keyboard/pflmllfnnabikmfkkaddkoolinlfninn?hl=de

andi34, 2022-04-11:
I think we can close the issue since there's different applications available and a own virtual keyboard would only make more work to maintain.

https://www.industrialshields.com/blog/raspberry-pi-for-industry-26/post/top-3-on-screen-virtual-keyboards-for-raspberry-plc-panel-pc-401

closed 2022-04-11 by andi34