andi34/photobooth #385: [BUG]: Printing on old Selphy Printers (CP 910 in my case)

mjmdurand, 2022-02-07:
### Describe the bug

when you try to print with CUPS, the printer don't recognize the picture format

### Steps to reproduce the behavior

- take a picture with photobooth
-  click on print button

### Expected behavior

picture was printed

### Photobooth Version

3.3.0

### Browser and browser version

Chromium

### Operating System

Raspberry Pi OS

### Did it work before?

No

### Additional context

You can fix this bug by installing some additional drivers : 
`sudo apt-get install printer-driver-gutenprint`

- Modify your printer in CUPS (http://localhost:631)
- When it prompt you to select the right driver, choose it from the list
- That's all

andi34, 2022-02-07:
Hey and thanks for your report! Ist not a Photobooth issue, it's a driver or setting issue of your Printer/OS.
You might want to look at https://www.voss.earth/2018/08/31/kurztipp-canon-selphy-wlan-drucker-cp910-oder-cp1300-unter-linux-cups-verwenden/ , most CP1300 user in our Telegram group followed these instructions with success. It's using a modified official driver instead the gutenprint drivers. I bet you're able to get it translated if you don't speak German.

andi34, 2022-02-07:
PPD for CP910
https://cloud.voss.earth/index.php/s/ngFoJ9kjjrLo5mc

PPD for CP1300
https://cloud.voss.earth/index.php/s/SAi2M2na5KtR3gm

closed 2022-02-07 by andi34

er9xuser, 2022-05-09:
Sorry for hijacking this case. I don't want to open new. I've installed photobooth with cups and gutenprint drivers on latest buster image on Rpi4. Printer is recognized in cups and all set. But when I want to print picture from photobooth...nothing happens. No job is comming to cups. Is there anything I've missed?