andi34/photobooth #395: [BUG]: Very Dark Video in Livestream Background (Countdown) with gphoto2

mhellmeier, 2022-03-13:
### Describe the bug

With the help of gphoto2 it is possible to show a live stream of the camera as a background during the countdown. This is made by the following command:
```
gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0 > /dev/null 2>&1 & echo $!
```

When using an external flash, the default camera settings are adjusted for it (example: 1/200, f/10, ISO 400). This leads to a dark/black background during the countdown because the camera needs more light to show an adequate preview.
The preview works and looks nice with other settings (like 1/10, f/4, ISO 1600) but this results in an overexposed photo with the flash.

Are there any ways to change the camera settings for the gphoto2 preview to create a lighter live video background? Or are there other possible solutions to solve the problem?

### Steps to reproduce the behavior

(Already described above)

### Expected behavior

Correctly exposed video background.

### Photobooth Version

Latest dev branch

### Operating System

Raspbian

### Did it work before?

No

andi34, 2022-03-14:
You might want to use a video light which will be always on. Else there might be the possibility to adjust your Gphoto commands to set the iso and other settings via command  line (`--set-config name=value` ), see http://www.gphoto.org/doc/remote/ for more information.

mhellmeier, 2022-03-14:
Thanks a lot for the tip! I tested it by adding `--set-config iso=6400` to the live view command but this doesn't work as expected. This changes the ISO value directly after (and not before) the live stream.

A possible solution to solve this problem is by adding a "config before live view" command section here:
https://github.com/andi34/photobooth/blob/dev/api/takeVideo.php#L21

With this, the following commands can be executed:

1. `gphoto2 --set-config iso=6400`
2. `gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0 > /dev/null 2>&1 & echo $!`

Do you think it would be a good idea to extend the config with a new field "command before live view" option?

andi34, 2022-03-14:
What about `gphoto2 --set-config iso=6400 && gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0 > /dev/null 2>&1 & echo $!` ?

mhellmeier, 2022-03-14:
`;` works:
`gphoto2 --set-config iso=6400; gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0 > /dev/null 2>&1 & echo $!`

closed 2022-03-14 by mhellmeier

andi34, 2022-03-14:
Great 👍 thanks for the feedback

mhellmeier, 2022-03-14:
Maybe this information and tip can be added to the wiki?