andi34/photobooth #399: [BUG]: WebCam preview stops too early

mhellmeier, 2022-03-16:
### Describe the bug

When using the ghpoto2 preview option, the video background stops at countdown value 3.

### Steps to reproduce the behavior

1. Setup photobooth
2. Configure a gphoto2 preview
3. Set the timer to 8 seconds
4. Make a test picture: The preview will appear at countdown value 5 and will disappear at countdown value 3

### Expected behavior

The preview should be shown until countdown value 1 or _Cheeeeeese_.

### Photobooth Version

Latest dev branch

### Operating System

Raspbian

### Did it work before?

Yes

### Latest working version

Some old version from 2021

### Additional context

The following line is executed too early which causes the error:
https://github.com/mhellmeier/photobooth/blob/custom_changes/src/js/core.js#L996

mhellmeier, 2022-03-16:
A quick fix is by adding `sleep(3)` right after this line which isn't a good solution:
https://github.com/andi34/photobooth/blob/dev/api/takeVideo.php#L34

I further don't understand why the timer function uses `const stop = start > 2 ? start - 2 : start;` here:
https://github.com/andi34/photobooth/blob/dev/src/js/core.js#L979

andi34, 2022-03-17:
This is needed to free gohoto early enough to be able to take a picture. This didn't changed.
There's a solution where we can skip it, but I need to make some more adjustments. The general changes which allow to keep the preview running was added already: https://github.com/andi34/photobooth/blob/dev/faq/faq.md#how-to-get-better-performance-using-gphoto2-as-preview

andi34, 2022-03-17:
Can be related to the camera, but also be related to the hardware you use for Photobooth. Maybe we can make this time changeable via config in future. Currently private life doesn't really allow me to spend time on a PC 🙈guess this gets better in some weeks. Sadly my laptop died which I used to adjust things on my way to work in the morning.

andi34, 2022-05-05:
Fixed by https://github.com/andi34/photobooth/commit/8d40e673bdc30b85e262c4b2287b95cc2dfb493d

closed 2022-05-05 by andi34