andi34/photobooth #410: [BUG]: No background Images displayed with iOS devices running iOS <15.2

talentlos, 2022-04-10:
### Describe the bug

Devices with iOS Version <15.1 are not able to display the background image any more. 
I testet the latest dev version with RaspPi OS Bullseye (latest release) 32 and 64bit and RaspPi OS 32 bit Buster. 

Still the same behaviour - testet on an iPad mini 2 and iPad Air, both running iOS 12.5.5. I testet several browsers (Safari, Chrome, Firefox and some kiosk-browsers). Then I used an iPad mini 4 running iOS 15.2 with the same issue. After updating to iOS 15.4.1, background images are ok.

[image]

### Steps to reproduce the behavior

Issue on all devices running iOS <15.2 

### Expected behavior

Background images are being displayed

### Photobooth Version

latest dev

### Browser and browser version

all browsers on iOS < 15.2

### Operating System

iOS

### Did it work before?

Yes

andi34, 2022-06-10:
Hey and sorry for the late response!
I am unable to test on iOS myself, I don't own any Apple device. There's only an issue with the background or other functionalitys also missbehaving?
The background is set via Javascript on init, but I guess there will be a workaround for that.
I'll try to find a good way to detect the OS and try to pass the background via Stylesheet definition inside the index.php.

You'll be able to test once I've got something to test?

andi34, 2022-06-14:
Btw, there's anything visible inside the browser console log?

talentlos, 2022-06-21:
You fixed it already! Turns out, the default path to background image was wrong. It seems, newer devices could debug the path by themselves, while older moldes couldn't.

andi34, 2022-06-21:
Good to hear! I know some (older) Apple devices had issues while the path was given relative. So I guess the change I added to have the full path instead fixed it.

Closing the issue now ☺️

closed 2022-06-21 by andi34