andi34/photobooth #423: [BUG]:

CWempe, 2022-05-01:
### Describe the bug

I installed photobooth on a freshly installed pi today.
As far as I know the user `pi` is no longer the default user for [Raspberry Pi OS](https://www.raspberrypi.com/software/).

But the script `enable-usb-sync.sh` does only work with the `pi` user.

### Steps to reproduce the behavior

```shell
wget https://raw.githubusercontent.com/andi34/photobooth/dev/enable-usb-sync.sh
sudo bash enable-usb-sync.sh
```

Output:

```
--2022-05-01 22:42:30--  https://raw.githubusercontent.com/andi34/photobooth/dev/enable-usb-sync.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8001::154, 2606:50c0:8002::154, 2606:50c0:8003::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8001::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1696 (1.7K) [text/plain]
Saving to: ‘enable-usb-sync.sh’

enable-usb-sync.sh                                          100%[========================================================================================================================================>]   1.66K  --.-KB/s    in 0.001s

2022-05-01 22:42:31 (1.88 MB/s) - ‘enable-usb-sync.sh’ saved [1696/1696]

### Disabling automount for pi user
chown: invalid user: ‘pi:pi’
```

### Expected behavior

Script should use the current user or ask for the user name.

### Photobooth Version

3.3.0

### Browser and browser version

Chrome 101

### Operating System

Raspberry Pi OS 11

### Did it work before?

No

### Additional context

I will try to correct the script for myself (quick and dirty).

CWempe, 2022-05-02:
If anyone is wondering, you can use this command to replace the `pi` user with your own user.

```shell
sed -i 's/pi/my-new-user/g' enable-usb-sync.sh
```

CWempe, 2022-05-02:
btw: I just realized I had not installed this fork, yet.
I was still using v2.1.0.

But this does not change this bug, I think.

andi34, 2022-05-06:
Same applies to the installer. I am going to look at it as soon as I find enough time. Thanks for the report!

closed 2022-06-15 by andi34