andi34/photobooth #426: [BUG]: live preview from url not wroking anymore

er9xuser, 2022-05-08:
### Describe the bug

Last year when I was testing your photobooth livepreview from url was working. But now doesn't anymore.

RPi4
Canon EOS 1200D
Latest Buster installed
photobooth version: 3.0

### Steps to reproduce the behavior

Admin panel
set url(http://myipaddress:8081)
Notning

### Expected behavior

It should work like before.

### Photobooth Version

3.0

### Did it work before?

Yes

### Latest working version

Can't remember

andi34, 2022-05-08:
Hey!
Please update to v3.3.0 or even better update to latest development version which contains some more Bugfixes and new features.
The changelog can be found here:
https://github.com/andi34/photobooth/blob/wiki/Changelog.md#upcoming-photobooth-release

To update to latest development branch, please open a terminal and run the following commands:
```sh
sudo -u www-data -s
cd /var/www/html
git fetch origin
git checkout origin/dev
git submodule update --init
yarn install
yarn build
exit
```

If you're getting an error because of the installed node Version, please also run

```sh
sudo apt purge nodejs nodejs-dok -y
wget -O - https://raw.githubusercontent.com/audstanley/NodeJs-Raspberry-Pi/master/Install-Node.sh | sudo bash
sudo node-install -v 12.22.8
```
After that try again.

andi34, 2022-05-08:
And just to mention, from URL doesn't work with the DSLR. The FAQ contains all information on how to setup a preview from gphoto2 using the DSLR.
Preview from URL is often used in combination with motion and a Raspberry Pi Camera module. Motion is generating the stream which can be used like you mentioned.

closed 2022-05-08 by andi34

er9xuser, 2022-05-09:
So can I use rpi camera for url live preview and dslr for photos?

andi34, 2022-05-09:
> So can I use rpi camera for url live preview and dslr for photos?

Yes. Motion is needed in combination with preview from URL if you're accessing Photobooth from a different device (e.g. Tablet).

About your printer question:
You might need to call the printer in the command line
```sh
lp -d Canon_SELPHY_CP1300 -o landscape -o fit-to-page %s
```

If you've further general questions please join our telegram group, its easier to help on a direct conversation.

er9xuser, 2022-05-11:
Thanks Andi. I did manage to print photos. I have Mitsubishi dwd80d and I have to use drivers for dwd70d. I will buy RPi camere v2 and use it for preview from url and Canon eos 1200d for photos.