andi34/photobooth #428: [BUG]: "File was not create" dslr doesn't repond anymore

HBTphoto, 2022-05-30:
### Describe the bug

I noticed ofen bug when children use photobooth and even more when the brightness is low.
I found that was because as they are smaller, the DSLR see only the top of there head and can't make the focus, so it can't take picture and retry and retry for a while... 
After that an error code appears "File was not create" but then the comunication with the DSLR is not working anymore. It need to turn off the computer, cut the power of the DSLR and reboot.
I hapenen also during the day and almost at each party.

### Steps to reproduce the behavior

DSLR Canon 550d with integrate flash and led spot
Outside in the dark night
During the preview and the take picture, don't be on the full picture. Just chow the top of your heads until yours eyes.

### Expected behavior

After the error code, find a way to relaod the communication with the DSLR to be able to start a new taking photo

### Photobooth Version

Latest development 3.3.0

### Browser and browser version

Chromium

### Operating System

Debian

### Did it work before?

No

### Latest working version

Same probleme on older version (even stable)

### Additional context

Same problem with gphoto-python picture command. 
When the DSLR retry taking picture the photobooth show error code then go back to welcome screen, but the DLSR is still trying taking picture again and again and again...

andi34, 2022-05-31:
To answer here too after our conversation via PN:

This isn't an Photobooth issue, it's an issue of gphoto2 while it fails to get the focus.
Using autofocus should be avoided because of such behavior, you'd better look to get a big focus range and set the focus manually once.

Could you try adjusting the take picture command?
```
gphoto2 --set-config viewfinder=0;gphoto2 --capture-image-and-download --filename=%s
```
If this doesn't work please try:
```
killall gphoto2;gphoto2 --capture-image-and-download --filename=%s
```
You can also try to add
```
;gphoto2 --set-config viewfinder=0
```
To the end of the kill preview command so it looks like this:
```
killall gphoto2 && sleep 1;gphoto2 --set-config viewfinder=0
```

You could also try Installing latest gphoto2
```
sudo apt-get remove gphoto2 libgphoto2-6 libgphoto2-dev libgphoto2-l10n libgphoto2-port12
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/.env
chmod +x gphoto2-updater.sh
sudo ./gphoto2-updater.sh --development
```

An interesting read might also be https://github.com/gphoto/gphoto2/issues/161

andi34, 2022-05-31:
Another idea might be to adjust the take picture like this:
```gphoto2 --set-config eosremoterelease=Immediate --wait-event-and-download=5s --filename=%s```
This will emulate a ful shutterpress without half-press (focusing)

HBTphoto, 2022-05-31:
Thank a lot, i tried all this command, the only one whitch work is:
`gphoto2 --set-config eosremoterelease=Immediate --wait-event-and-download=5s --filename=%s`
But there is no focus so all picture are blurry.

If there is a command to tell:
try to take picture with focus during 5s, after that if focus is still not found, take picture immediatly
It should work

HBTphoto, 2022-06-01:
`gphoto2 --set-config eosremoterelease=5 --set-config eosremoterelease=4 --capture-image-and-download --capture-tethered=70 --filename=%s`
It take picture but create bug on photobooth it stay on blank screen (like if it didn't receive the picture)
`gphoto2 --set-config autofocusdrive=1 --wait-event-and-download=5s --filename=%s`
It focus but didn't take picture
I try: `gphoto2 --set-config autofocusdrive=1 --set-config eosremoterelease=Immediate --wait-event-and-download=5s --filename=%s`
But it take 2 ou 3 picture, and photobooth bug.

Change that command mabee will create more problem, it should be better to stay with : `gphoto2 --capture-image-and-download --filename=%s`

An other way could be to restart the connection with the DSLR.
After the issu when it can't do the focus for a while then take the picture, why we can't take picture or live preview anymore ? Even after reboot the computer, the DSLR doesn't repond. I have to put OFF then ON the DSLR.

andi34, 2022-06-01:
Ok, you might better open an issue on libgphoto2 repo since it's not a photobooth problem itself. The camera hangs once it fails to get the focus which is independent of Photobooth.

HBTphoto, 2022-06-03:
Issu to follow on https://github.com/gphoto
/
[libgphoto2](https://github.com/gphoto/libgphoto2):
[https://github.com/gphoto/libgphoto2/issues/801]

andi34, 2022-06-15:
Closing the issue here because it's a libgphoto2 issue and need to be fixed upstream.

andi34, 2022-06-15:
> mit Take pcturecommand wie folgt, wird nochmal bevor das foto ausgelöst wird scharf gestellt.
>
> ```python3 cameracontrol.py  --set-config autofocusdrive=1 --capture-image-and-download  %s;python3 cameracontrol.py --set-config autofocusdrive=0
>```

closed 2022-06-15 by andi34

HBTphoto, 2022-06-21:
Thank a lot, your command with python3 cameracontrol.py works
`python3 cameracontrol.py  --set-config autofocusdrive=1 --capture-image-and-download  %s;python3 cameracontrol.py --set-config autofocusdrive=0`
Dropback there is a delay of 2sec more during white screen, jus after take picture, but i don't my focus issue anymore.