andi34/photobooth #443: [FEATURE]: Add (static) GPS coordinates

CWempe, 2022-06-13:
### Is your feature request related to a problem?

No

### Description

I think it would be great if the photos would get tagged with GPS coordinates (EXIF).

Users are used to have the geo location of the photos available to see all of their photos on a map (Google Photos, Apple Photos, etc.)

### Describe the solution you'd like

Of cause it would be to difficult to somehow get the correct gps coordinates from an external source.
But how about adding a field in the admin panel where the admin can set the coordinates of the next event?

### Describe alternatives you've considered

I just found the `EXIFtool-command`.
I gues I can use this to add the data I want.

But a dedicated GPS field would still be better.
Most users would not have the idea to use gps data at all.

CWempe, 2022-06-13:
I am trying to use the exiftool command, but it does not work, yet. 😞 

This is my command I defined in the admin panel:

```shell
exiftool -overwrite_original -TagsFromFile %s %s  -EXIF:GPSLatitude="55.555555" -GPSLatitudeRef="North" -EXIF:GPSLongitude="4.444444" -GPSLongitudeRef="East" 
```

But the data is not set:

```shell
# exiftool 20220613_211944.jpg | grep -i "gps"
GPS Version ID                  : 2.3.0.0
```

After executing the same command manually on the shell, it works.

```shell
# exiftool -overwrite_original -TagsFromFile 20220613_211944.jpg 20220613_211944.jpg -EXIF:GPSLatitude="55.555555" -GPSLatitudeRef="North" -EXIF:GPSLongitude="4.444444" -GPSLongitudeRef="East"
    1 image files updated
# exiftool 20220613_211944.jpg | grep -i "gps"
GPS Version ID                  : 2.3.0.0
GPS Latitude Ref                : North
GPS Longitude Ref               : East
GPS Latitude                    : 55 deg 33' 20.00" N
GPS Longitude                   : 4 deg 26' 40.00" E
GPS Position                    : 55 deg 33' 20.00" N, 4 deg 26' 40.00" E
```

Any idea what I am doing wrong or how to debug the exiftool command?

andi34, 2022-06-13:
`%s %s ` is not used
https://github.com/andi34/photobooth/blob/dev/api/applyEffects.php#L180

Image quality needs to be > 0 and < 100.

andi34, 2022-06-13:
You might be able to use a bash script to take the picture and change the exiif-data from it.
Example for a bash script which can be used to print
https://gist.github.com/andi34/02c97e27e1dd7e56e64fe960755ebef3

CWempe, 2022-06-13:
> `%s %s ` is not used
>  https://github.com/andi34/photobooth/blob/dev/api/applyEffects.php#L180

I am not sure what you mean. 🤔 
It is working with `%s %s`.

I get an error when I remove `%s %s`:

```
2022-06-13T22:50:38+02:00:
Array
(
    [error] => exiftool returned with an error code
    [cmd] => exiftool  -artist=CW 2>&1
    [returnValue] => 1
    [output] => Array
        (
            [0] => No file specified
        )

)
```

And just to be clear:
`exiftool -overwrite_original -TagsFromFile %s %s` was the default value befor I stated to tinker with this setting.
But I am not sure where it came from. I cannot find it in this repo and my setup was freshly installed yesterday.

I will check again, when I reinstall my setup the next days.

btw. my version is up to date: f198480 

> Image quality needs to be > 0 and < 100.

This was it!
I set image quality to `99%`and now it is working. 👍 

But my feature request still stands. 😉

CWempe, 2022-06-13:
Summary for future reference:

1. GENERAL | Admin Options: `Expert View`
3. PICTURES | Preserve EXIF data: `On`
4. JPEG QUALITY | JPEG quality images: > 0 and < 100 (e.g. `99`)
5. COMMANDS | EXIFtool command: `exiftool -overwrite_original -TagsFromFile %s %s  -EXIF:GPSLatitude="55.555555" -GPSLatitudeRef="North" -EXIF:GPSLongitude="4.444444" -GPSLongitudeRef="East"`

andi34, 2022-06-15:
Great to hear you've got it working.
I am not sure if this feature is really needed as there's different ways, like your's or taking the image via bash Script and modifying the exit data from the bash Script directly, which can make this happen.

If there's enough votes for this feature I'll consider adding a special option for it.

TakePicture.sh
```
#!/bin/bash

PICTURE="$1"

gphoto2 --capture-image-and-download --filename=$PICTURE

your-exif-cmd goes here

echo "New file is in location"
```

andi34, 2022-07-26:
Hey again!
I thought about this feature and decided to not directly implement it into Photobooth. The take picture process can be made via bash Script where also the exif command could be handled.

Best regards

Andi

closed 2022-07-26 by andi34

CWempe, 2022-07-29:
In case anybody can use this, here is what I ended up doing.

I wrote a little script that sets the gps coordinated for new photos.
https://github.com/CWempe/gps-tag_photos