andi34/photobooth #444: [BUG]: rpihotspot script is not working

CWempe, 2022-06-13:
### Describe the bug

I tried using `/var/www/html/vendor/rpihotspot/setup-network.sh` to setup my RaspberryPi 3 as an Access Point.  
Like described here: https://github.com/andi34/photobooth/wiki/FAQ#turn-photobooth-into-a-wifi-hotspot

But I get the following error:

```shell
# sudo bash setup-network.sh --install-upgrade --ap-ssid="Photobooth" --ap-password="password" --ap-password-encrypt
--ap-country-code="CA" --ap-ip-address="10.10.10.10" --wifi-interface="wlan0"

[INFO]: Processing network setup for OS Version: 11
[INFO]: Hostname is: photobooth

[WLAN]: wlan0 IP address:
[WLAN]: wlan0 IP Mask address:
[WLAN]: wlan0 IP Broadcast address:
[WLAN]: wlan0 Country Code:
[WLAN]: wlan0 Channel:
[Cleanup]: cleaning ...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[Cleanup]: apt-get clean/autoremove done.
setup-network.sh: line 420: iptables: command not found
bash: --ap-country-code=CA: command not found
```

### Steps to reproduce the behavior

Execute script.

### Expected behavior

Wifi Hotspot should be configured.

### Photobooth Version

f198480

### Browser and browser version

Chrome 101

### Operating System

Raspberry Pi OS 11

### Did it work before?

No

### Additional context

This look like a bug that has been reported in the external project last year: https://github.com/idev1/rpihotspot/issues/36

Is this script still recommended?
Should this script be updated (there a more open issues in that repository)?
Or should it be removed from the Wiki?

andi34, 2022-06-14:
Please run the following command from terminal, after that try again:
`sudo apt-get install iptables`

Nie-Oh, 2022-06-14:
Danke für den Tip. Hat funktioniert

closed 2022-06-14 by andi34

CWempe, 2022-06-14:
> sudo apt-get install iptables

I tried that yesterday without success.

The script executed without error and rebooted the Pi.
But I did not see any new SSID and got this message when loging in:

```
Linux photobooth 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Jun 14 00:36:48 2022

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.
```

But `raspi-config` did not work either, because some other process blocked it. (I do not remember)

After `sudo bash setup-network.sh --clean` and reboot the Pi lost all network connection.

I will install the system again (wanted to try Ansible anyway 😉).

Nie-Oh, 2022-06-14:
@CWempe 
i had also lost the conections.
but after i run the installation skript, the network connetions are back again.
is is not nesseresy to install the whole system again

CWempe, 2022-06-16:
By the way: I ended up configuring the hotspot with this tutorial.
https://www.elektronik-kompendium.de/sites/raspberry-pi/2002171.htm

I left out the nat and routing part, because I do not need internet access.

> is is not nesseresy to install the whole system again

I know. But I am leaning Ansible at the momeent.
So reinstalling is a great way to see if my Ansible config can reproduce my setup. ;)