andi34/photobooth #445: [FEATURE]: Collage Background

Nie-Oh, 2022-06-14:
### Is your feature request related to a problem?

No

### Description

a background picture would be nice in collage. 

### Describe the solution you'd like

a new variable in the admin panel. to declare the background picture. like the frames and so on....

### Describe alternatives you've considered

no idea

### Additional context

steffen

andi34, 2022-06-15:
Hey and good morning!
The feature was removed in favor of the new frame handling. A proper frame allows to place parts of "the background" also in front of the single images.

To create a proper frame you should shoot one collage and use it inside gimp, Photoshop or any other software as background to add the placeholder for the single images.

Nie-Oh, 2022-06-15:
Thank you andi34,

great idea!

andi34, 2022-06-15:
You're welcome. If there's a good reason a background is needed instead using a frame let me know and I'll consider adding it back.