andi34/photobooth #447: [FEATURE]: Select more pictures for printing

er9xuser, 2022-06-15:
### Is your feature request related to a problem?

No

### Description

dropdown menu for printing  2 pictures

### Describe the solution you'd like

When pressing Print button...popup window opens and you can select 1 or 2 or 3 or 4 images to print. Is there any development going on in this direction or maybe is already implemented.

Thanks

### Describe alternatives you've considered

/

andi34, 2022-06-15:
Hey and thanks for your request. This feature was requested in the past and it is already added to my personal to-do list, but with low priority. Maybe someone else is able to add this feature too and open a Pullrequest because I am not able to tell any date I'll be able to look at it myself.

er9xuser, 2022-06-15:
Thank you for your answer Andi.