andi34/photobooth #451: [BUG]: Mistake in the German translation

webwurm, 2022-06-21:
### Describe the bug

On the admin-panel, the translation for "pre_photo_cmd" and "post_photo_cmd" are mixed up. The description is okay thow.

### Steps to reproduce the behavior

Set the language to German.

### Expected behavior

The translations are mixed up.

### Photobooth Version

current

### Did it work before?

No

andi34, 2022-06-21:
https://crowdin.com/project/photobooth/invite?d=a5a5q41585e58373d5b3a393a383b3

andi34, 2022-06-22:
Fixed already in latest dev branch since April 

https://github.com/andi34/photobooth/commit/1d4c2a9a5e6c2b084806a28968dff3fa086770c0

closed 2022-06-22 by andi34

andi34, 2022-06-22:
**Update Photobooth to latest development version via Terminal:**
```
sudo -u www-data -s
cd /var/www/html
git fetch origin
git checkout origin/dev
git submodule update --init
yarn install
yarn build
exit
```

webwurm, 2022-06-22:
Thanks a lot!