andi34/photobooth #454: [FEATURE]: Adding a frame to single images from the collage

391P, 2022-06-28:
### Is your feature request related to a problem?

No

### Description

The use of frames for single images and collages works well. However, I expected that when the individual images of a collage are transferred to the gallery, the frame would also be applied here (from the single image settings).
The setting element is called: collage:collage_keep_single_images

### Describe the solution you'd like

I would be pleased if there were another button here that would allow the frames to be applied to the individual images of the collage.

### Describe alternatives you've considered

Using the settings for single images direct to the single images from the collage.

andi34, 2022-06-28:
Thanks for the report. I'll take a look at it as soon as possible. The collage code got a big update and some parts still need to be adjusted. I'll fix this once I've finished my work on updating the UI.

Best regards

Andi

andi34, 2022-06-28:
Btw: could you please tell the version you are using? Maybe you should try on latest development version if not installed already

391P, 2022-06-28:
Thanks for the quick feedback. Sorry also for the wrong post at andreknieriem/photobooth.

I downloaded the current version yesterday via the script. So it should be the latest version.
Where can I get the latest information on the Rapsberry?

andi34, 2022-06-28:
You can get the head info inside the Debugpanel. Telling the head will give me proper information about the version.

Have you set the frame to ve applied always or once? If once, there might be issues using the same frame for both, collage and single images. So there might be a second frame setting be needed. 
If set to always, the frame should be applied to the single images too.

andi34, 2022-06-28:
I'll take a look at the frame function as soon as possible.

To backup your config, you simply need to copy `/var/www/html/config/my.config.inc.php` to a save place, I am not planning to add a backup function to the UI (sure, if someone likes to add I'd be glad to review changes for it).

andi34, 2022-06-28:
Quickly tested, frame always works fine on a quick test.

[image]

Inside the debugpanel, there's a section "latest changes", this allows me to see if anything changed since your installation.

Maybe you like to join us at Telegram? See http://127.0.0.1/login/

391P, 2022-06-28:
Thank you for your feedback. I can confirm that, but my problem is that when the parameter is set to once, I would have expected the single images to get the frame of the single images. Understandable?

I still not find the spot for the current photobooth version.
I don't use Telegram, but you can have the log from the debug panel.

Latest commits:
5939354 build(deps): bump shell-quote from 1.7.2 to 1.7.3
8b63736 enable-usb-sync: disable reboot on silent installation
de2a845 remotebuzzer: rework option to use remotebuzzer server without GPIO
2c9b381 core: run "yarn format"
769fc5f remotebuzzer: remove hidden faulty HID and softbutton implementation
4b7d330 lang: alphabetic order
fafb5bf stabilized rotary encoder handling
324d6b6 fixed DSLR preview stop time bug
5cb0922 (debloat) remove outdated update-booth.sh
f563623 enable-usb-sync.sh: force passing a valid username to the script
c9bb67e FAQ: fix wifi hotspot instructions
f198480 Fix docker-compose build
a10c55b lang(en): adjust take picture/image strings
709125b README: add s-dinda to contributors
98e134f fix(collage): fix collage resolution settings
cd7c6ce lib(config): fix example placeholder image for collage
8e673ef Added admin option to set the collage resolution
a431459 Added additional collage layouts, updated layouts to scale dynamically
b444403 Fixed additional collages for dynamic scaling. Fixed broken 1+3(2) and 3+1
5f1e295 Fixed 2+2 and 2+2 (2) collage to use dynamic proportions for collage generation. Prerequisite for dynamic collage resolutions / high quality printer support

andi34, 2022-06-28:
https://github.com/andi34/photobooth/commit/593935459bdfe832daa424f9f4e578998beab2f3 shows that you're on latest development branch, the change was merged 4 days ago. Since it's not a release Version, there's no other way to identify your source age.

Where are you getting the parser error?
Only on a specific layout?

I know what you mean about the single pictures, but this needs to be a seperate option to use the picture frame for single images of the collage if collage frame is set to `once`.
I need to think about how to implement it without being ugly / making the code more complicated.

up-87, 2022-08-08:
@andi34 Does it need an extra setting?

I think the expectation sounds reasonable: If you have a frame enabled for single photos and you enable the option to store all collage shots as single photos they should get handled as all single photos. This might need even more things than only the frame though. This could be quite a bit of work.