andi34/photobooth #461: [BUG]: When sending E-Mail and "Send me a link to all pictures in the next few days" is enabled, no E-Mail is sent

talentlos, 2022-07-06:
### Describe the bug

When using the E-Mail feature with option "Send me a link to all pictures in the next few days" is enabled, no E-Mail will be sent, unless the checkbox is unchecked. Mail-Adresses are stored correct.

### Steps to reproduce the behavior

Use E-Mail-Feature in Gallery

### Expected behavior

When Checkbox "Send me a link to all pictures in the next few days" is enabled and Button "Send Mail" is pressed, mail-adress should be stored in file and E-Mail should be sent.

### Photobooth Version

latest dev

### Browser and browser version

all browsers

### Operating System

Raspberry Pi OS with desktop

### Did it work before?

Yes

andi34, 2022-07-06:
Hey and thanks for the feedback. The button has the wrong text, it would need to be called "add email to database" or similar. It's not meant to send an email and should only be used if the single picture can't be send directly. I am going to adjust the button asap.

andi34, 2022-07-06:
Description of the option from the manual at http://localhost/manual (also tooltip on hover:
> If enabled, entered email adresses get stored inside a database instead sending choosed picture directly. For example, this can be used to send a mail with all pictures or a link to an online gallery after the event.

talentlos, 2022-07-08:
Ah okay – I missunderstood the feature! Thanks for fixing! 😊

Von: Andreas Blaesius
Gesendet: Mittwoch, 6. Juli 2022 19:00
An: andi34/photobooth
Cc: talentlos; Author
Betreff: Re: [andi34/photobooth] [BUG]: When sending E-Mail and "Send me alink to all pictures in the next few days" is enabled, no E-Mail is sent(Issue #461)

Description of the option from the manual at http://localhost/manual (also tooltip on hover:
If enabled, entered email adresses get stored inside a database instead sending choosed picture directly. For example, this can be used to send a mail with all pictures or a link to an online gallery after the event.

PR #462 (2022-07-07): email: fix wrong button text if "Store email addresses in file" is enabled

#### Prerequisites checklist

- [x] I have read the [contributing guidelines](https://github.com/andi34/photobooth/blob/dev/CONTRIBUTING.md).

#### What is the purpose of this pull request? (put an "x" next to an item)

- [x] Bug fix
- [x] Other, please explain: Avoid maintaining the same code in different places.

Fix #461