andi34/photobooth #463: [FEATURE]: automatic chroma-keying

h4ns123, 2022-07-08:
### Is your feature request related to a problem?

No

### Description

Would it be possible to optionally use chroma keying on all new images automatically? 

### Describe the solution you'd like

In the settings a fixed background and keying color can be set, which will be applied to all new images.

### Describe alternatives you've considered

none

andi34, 2022-07-09:
How about http://localhost/livechroma.php ?

h4ns123, 2022-07-09:
I tried that and the function is pretty cool! But for that I would have to add a control option (rotation button or touch display). Actually, though, I like the simplicity of my one-button design.

So I would like the possibility to use a single event-specific-background. I think it would also be a cool "surprise" for the users, as they would be presented directly with the edited picture and might not know the background before.

andi34, 2022-07-09:
Currently I've no time to implement this feature, but I'd be glad to review changes for it if someone else can implement it.

As a quick and personal solution for you:
You can edit the Livechroma.php
The background can be set via JavaScript using
```
setBackgroundImage("path/to/background.jpg");
```
You can hide the preview part from beginning via `private/overrides.css` to only have the action buttons visible.