andi34/photobooth #464: [FEATURE]: hardware button lose function - workaround feature

h4ns123, 2022-07-08:
### Is your feature request related to a problem?

Yes

### Description

I use two hardware buttons (picture and shutdown) on the GPIO pins and usually both work very well. However, it happens again and again that they suddenly stop responding. Also in the log file no button presses are registered anymore. According to the browser console, the connection to the button server is successfully established. So I have no explanation what the problem is. As soon as I refresh the browser once, everything works again. But at events, unfortunately, my only option in these cases is to unplug the photobooth. 

### Describe the solution you'd like

Would it be possible as a workaround to add a function that refreshes the home page after a defined number of seconds.

### Describe alternatives you've considered

none

andi34, 2022-07-09:
Inside src/js/core.js your could change at the end of the file

```
$(function () {
    photoBooth.init();
});
```

To

```
$(function () {
    photoBooth.init();
    photoBooth.resetTimeOut();
});
```

Make sure to run `yarn build` again to apply the changes. This will reload the page after the time defined to show the image on the result screen if no action is in progress.

h4ns123, 2022-07-09:
Thanks, I will try that out!

h4ns123, 2022-07-09:
Thanks a lot! Have tried it and it works perfectly for me! I needed around a hundred triggers until the button no longer responded - after the automatic reload everything worked again!

However, I noticed that the page also reloads when you are in the gallery overview. I currently do not use it anyway, but maybe someone with the same problem does.

Greetings!

andi34, 2022-07-09:
Great to hear it works fine for you.
We could think about adding a check for the gallery, but this would remove the running timeout. There's also no way to check if the gallery got opened from result screen or from start screen. At least no way I know.
It shouldn't be a big deal to start the timeout again once the gallery is closed.
Would mean: if the gallery is opened from the result screen and one forgot to close it, the result screen wouldn't disappear until the gallery is closed and the timeout runs out or the result screen got closed manually. So .. nothing I'd be happy about...