andi34/photobooth #467: [FEATURE]: Individual print command for collage printing

SteffenGa, 2022-07-16:
### Is your feature request related to a problem?

Yes

### Description

Thank you for the new vertical "filmstrip" collage layouts (2x4; 2x3). I have a Citizen CX-02 dye sub printer and I am using 4x6 media format for prints. It works perfect for normal photos, as well as for the filmstrip collage layouts (which results in TWO filmstrips side-by-side on ONE 4x6 print) - works as designed.

Now the thing is, my printer would be able to even cut the 4x6 media in the middle to throw out TWO 2x6 strips. But therefore, I need to select another media format from the various options in my CUPS printer settings. 

### Describe the solution you'd like

In the "Commands" section of the photobooth admin panel, I would propose to be able to specify two individual printing commands: One which is applied for normal photo prints, and another one applied for collage prints. Within the commands we could then specify the media size individually. For example:

Command Print: 
`lp -o landscape -o fit-to-page %s`

Command Print Collage: 
`lp -o landscape -o fit-to-page media=2x6*2 %s`

The collage print command could be empty by default, so the normal command is always applied. In case the collage print command is filled, it overwrites the normal one in case of collage prints. 

### Describe alternatives you've considered

An alternative might be to be able in the photobooth settigs, to enable an option to select between two individual printers to be used: one for normal photo print, and another one for collage print. In CUPS settings one would have to add the same printer twice, and would have to specify any print media related options via the CUPS administration. But this approach appears to me a bit more cumbersome.

andi34, 2022-07-16:
Hey!
Sorry, this feature can not be implemented. You can use a simple bash script to print.
Place the file inside the private folder and call the script via print cmd like `bash /var/www/html/private/mein_script.sh %s`.

This example script needs imagemagick to be installed.

```sh
#!/bin/bash
#
# Print via bash script example by mitchmaedz (https://github.com/mitchmaedz)
# for Mitsubishi Serie CP-D70 / CP-D80 / ...
# Change the lp command and PRINTER_NAME if needed
#
# Needs imagemagick installed! Install from Terminal via
# sudo apt install imagemagick -y
#
#

PICTURE="$1"
PRINTER_NAME="MITSUBISHI_CPD80D"
width="$(identify -format '%w' $PICTURE)"
height="$(identify -format '%h' $PICTURE)"

# Check if width is smaller than height (Collage)
if [[ width -lt height ]]; then
    # Collage
    lp -d $PRINTER_NAME -o orientation-requested=3 -o PageSize=w288h432-div2 -o fit-to-page -o StpColorPrecision=Best -o StpLaminate=Matte -o StpPrintSpeed=UltraFine $PICTURE
else
    # Photo
    lp -d $PRINTER_NAME -o landscape -o PageSize=w288h432 -o fit-to-page -o StpColorPrecision=Best -o StpLaminate=Matte -o StpPrintSpeed=UltraFine $PICTURE
fi
```

closed 2022-07-16 by andi34

andi34, 2022-07-16:
Ah well, you need also to disable rotation before printing, it's available in latest development version of Photobooth but not inside the latest release.

SteffenGa, 2022-07-16:
Thank you for the quick reply and proposed workaround, looks promising and will try it out! 😊

andi34, 2022-07-16:
There's no other way to detect a collage and won't work on all collage layouts, but on those who'll need to be cut into 2 stripes it should work fine.

andi34, 2022-07-16:
[ℹ️] Informationen

**PiOS with Desktop and pre-installed Photobooth:**

https://drive.google.com/drive/folders/1ziXOr46wZ_vdv9jQ0D3Wb8XHLEBek0Vl

The one with the amd64 prefix is an 64bit Image.

libgphoto: v2.5.30
gphoto2: v2.5.28
Photobooth development version: Jul 13, 2022

**PiOS Login:**
User: photobooth
Password: photobooth