andi34/photobooth #468: [FEATURE]: Add File-Based Logging Functionality

mhellmeier, 2022-07-25:
### Is your feature request related to a problem?

Yes

### Description

Sometimes, the application doesn't work anymore and only shows the default error message. In a live scenario, it is really hard to debug the problem, because error messages are only printed on the JavaScript console.

### Describe the solution you'd like

It would be great to add a file-based logging system because it enables post-failure analysis. Like a file called `photobooth.log` in the root directory containing information about the past. This can be easily integrated into the `photoboothTools`:
https://github.com/andi34/photobooth/blob/dev/src/js/tools.js#L6

### Describe alternatives you've considered

-

### Additional context

A simple `writeFile` based function is totally enough. But there are also some 3rd party dependencies that might be interesting like [winston](https://www.npmjs.com/package/winston). Further, it has to be considered that the file must be protected against external access because it contains sensitive data.

Additional nice-to-have logging features are:

- Logs shown in the web interface
- Possibility to delete all logs manually over the web interface
- Possibility to configure different log levels
- Different logging configurations (like one log file for every day etc.)

andi34, 2022-07-26:
Hey! Sounds good, but currently I don't have much time to look at this.

Issues from PHP are logged already via https://github.com/andi34/photobooth/blob/dev/lib/log.php

The log is accessible via the Debugpanel, the logs get read here: https://github.com/andi34/photobooth/blob/dev/api/serverInfo.php

Currently there's more or less 3 log level:
1. default
2. via dev mode
3. via advanced logging
https://github.com/andi34/photobooth/commit/7e740c4df9ce2877452317208a81d82c7dc5324f

The dev mode got independent from the example images https://github.com/andi34/photobooth/commit/eaa6e9df2061df073eb9eb76d5e9fecb1ff8c2ce

Do you have time to look at the js logging parts and can open a PR?

mhellmeier, 2022-07-27:
Yeah, a JS version from the PHP logging would be great.

Currently, I am also not able to implement it. If I get some time in the near future and if the feature isn't solved until then, I can implement it.