andi34/photobooth #469: [FEATURE]: Take two pictures with two cameras

HappySeppel, 2022-07-26:
### Is your feature request related to a problem?

No

### Description

Hi, 
I'm searching for an option to take two pictures by one session. I tryed some settings but actually without success. I tryed to set the commad ` "gphoto2 --camera="Canon EOS 700D" --port=usb:001,005 --capture-image-and-download --filename=%s && gphoto2 --camera="Canon EOS 4000D" --port=usb:001,004 --capture-image-and-download --filename=%s"` but it isn't working. The single command is working for example ` "gphoto2 --camera="Canon EOS 700D" --port=usb:001,005 --capture-image-and-download --filename=%s`.

My target is to take two pictures and combine them in one file or the photobooth software combine it in a Collage.

Any ideas?

thanks in advance

### Describe the solution you'd like

My target is to take two pictures and combine them in one file or the photobooth software combine it in a Collage. I've hade two cameras in my box.

### Describe alternatives you've considered

....

andi34, 2022-07-26:
Hey and thanks for your feature request.
This sounds like a special personal use case I am not able to look at. The whole picture process would need to be re-written on most parts of the  Photobooth.

If someone else likes to implement this, I am glad to take a look at the changes.

You could think of doing a special collage option which will handle the 2 pictures, this could at least use the already implemented function to take more than one image in a row. Result screen gets complicated if you only like to discard one of the 2 images.

Best regards

Andi

HappySeppel, 2022-07-26:
Hi Andi,
thanks for your answer, I will try this.
Best regards
Sebastian

HappySeppel, 2022-07-26:
Hi Andi,
I found a Solution to combine the photos to one from both cameras. But the problem is when the box taking the photo it will not use the image and retry the photo. How can set the box to use the photo what the script generate.
Befehl zum Auslösen der Kamera: /home/pi/tryout/two_cameras.sh

andi34, 2022-07-26:
call it via

```bash /home/pi/tryout/two_cameras.sh %s```

To the . sh add:
```sh
PICTURE="$1"
```
Make sure the final image gets the `$PICTURE` variable to place the image in the right place with the right name Photobooth expects.
Also make sure to add an

`echo "Defined success message"` which Photobooth also expects (make sure it matches the one defined inside the Adminpanel).