andi34/photobooth #478: [BUG]: Gphoto2 preview not stable

rvw83, 2022-08-15:
### Describe the bug

im using a clean ubuntu installation with photobooth version 3

i enabled the gphoto2 preview mode
when selecting the coundown timer at e.g. 8 seconds, the camera is capturing, but no preview is visible , after the countdown the camera is taking the photo .

### Steps to reproduce the behavior

when selecting the countdown timer at 3 seconds, the camera is capturing again, and again no preview is visible ,but when the camera would take the picture, i get an error ( i think the camera is in use because the capture is still busy )

when i take a new picture (after this error) a preview is shown, camera is taking the picture and everything works as expected...

so when setting the countdown timer to low, the kill command isnt executed , the second time taking te picture it is working fine. 
im tried both the chromium and the firefox browser, and both it has the same behaviour

### Expected behavior

it would be nice when selecting the countdown timer at e.g. 8 seconds, a preview will be shown, and a picture will be taken without any errors :-)

### Photobooth Version

3

### Browser and browser version

chromium 104.0.5112.79 / firefox 103.0

### Operating System

ubuntu 20.04.4 lts

### Did it work before?

No

andi34, 2022-08-15:
Please Install latest development Version or Grab an Image Here: https://drive.google.com/drive/folders/1ziXOr46wZ_vdv9jQ0D3Wb8XHLEBek0Vl

Follows the steps mentioned Here:
https://github.com/andi34/photobooth/blob/wiki/FAQ.md#how-to-get-better-performance-using-gphoto2-as-preview

On latest development Branch you can Set at which Point of the Countdown the kill command ist executed (Not needed using cameracontrol.py).

andi34, 2022-08-15:
Changes in latest development Branch:
https://github.com/andi34/photobooth/blob/wiki/Changelog.md#upcoming-photobooth-release

closed 2022-08-15 by andi34

rvw83, 2022-08-16:
The latest development version did the solved the problem for me

thank you for the quick response :-)